aderusha/HASwitchPlate #1: Fhem intergration

dj-bauer, 2017-09-07:
Cool Project, but could you write an fhem integration for it?

aderusha, 2017-09-07:
This project is platform-agnostic and relies only on MQTT. I'm providing demonstration automations for Home Assistant as that is what I personally use and it effectively shows how to interact with this project in a (semi)readable manner.  Creating specific integrations for every MQTT-enabled automation platform out there would be a huge undertaking and is out of scope for this project.

Having said all that: if you take a swing at it and post the results somewhere, I'll happily link to your work in the documentation for other users to find!

aderusha, 2017-09-07:
Issue closed - out of scope.

closed 2017-09-07 by aderusha