aderusha/HASwitchPlate #7: HASwitchPlate Connected automation triggers multiple times

squirtbrnr, 2018-04-03:
The HASwitchPlate_Connected automation triggers multiple (close to 15 times) when cycling power on the panel or when it reconnects to WiFi/MQTT

squirtbrnr, 2018-04-03:
file: HASwitchPlatePages.yaml

aderusha, 2018-04-18:
I can't seem to reproduce this behavior.  I've deployed a new instance of hass (currently `0.67.1`) and deployed the stock config from the repo here with the following command:
`svn export https://github.com/aderusha/HASwitchPlate/trunk/Home_Assistant ~/.homeassistant --force`

I started up Hass and configured the default state of the page selection buttons with the following commands:
```
curl -X POST -H "x-ha-access: YOUR_PASSWORD" -H "Content-Type: application/json" -d '{ "entity_id": "input_text.haswitchplate_pagebutton1_label", "value": "scenes" }' http://localhost:8123/api/services/input_text/set_value
curl -X POST -H "x-ha-access: YOUR_PASSWORD" -H "Content-Type: application/json" -d '{ "entity_id": "input_text.haswitchplate_pagebutton2_label", "value": "status" }' http://localhost:8123/api/services/input_text/set_value
curl -X POST -H "x-ha-access: YOUR_PASSWORD" -H "Content-Type: application/json" -d '{ "entity_id": "input_text.haswitchplate_pagebutton3_label", "value": "media" }' http://localhost:8123/api/services/input_text/set_value
curl -X POST -H "x-ha-access: YOUR_PASSWORD" -H "Content-Type: application/json" -d '{ "entity_id": "input_number.haswitchplate_pagebutton1_page", "value": 1}' http://localhost:8123/api/services/input_number/set_value
curl -X POST -H "x-ha-access: YOUR_PASSWORD" -H "Content-Type: application/json" -d '{ "entity_id": "input_number.haswitchplate_pagebutton2_page", "value": 2}' http://localhost:8123/api/services/input_number/set_value
curl -X POST -H "x-ha-access: YOUR_PASSWORD" -H "Content-Type: application/json" -d '{ "entity_id": "input_number.haswitchplate_pagebutton3_page", "value": 8}' http://localhost:8123/api/services/input_number/set_value
curl -X POST -H "x-ha-access: YOUR_PASSWORD" -H "Content-Type: application/json" -d '{ "entity_id": "input_number.haswitchplate_active_page", "value": 1}' http://localhost:8123/api/services/input_number/set_value
```
Once Hass has initialized I powered on the HASwitchPlate and observed all automations running once.  I power cycled the HASwitchPlate device several more times with Hass running and observed the same behavior each time.

Can you test this with a clean configuration by chance?  Is there any way another automation you've deployed might be interacting somehow?

aderusha, 2018-04-22:
I've narrowed the issue down to two compounding problems.  One is [an issue with Hass for which I have submitted a bug report](https://github.com/home-assistant/home-assistant/issues/14047).  With MQTT discovery enabled, hass is sending multiple replies in response to a single trigger.  This behavior is compounded by the page switching automations, which in certain cases can result in the page command being sent twice.  Because Hass is duplicating the automation response, the system can get stuck in a loop sending page commands back and forth, each of which triggers an additional cascade of MQTT messages until the Arduino code can't keep up and either resets or takes too long to respond to keep the loop going.  The result can be 10 or more responses on some of the page flipping automations.

squirtbrnr, 2018-05-06:
this is still occurring.  in fact I have the switch plate disconnected from power yet whenever I restart HA, the state goes back to "Connected" even though the switch plate is physically powered off, then the automation triggers now close to 25 times