aderusha/HASwitchPlate #10: WiFi autoconfig not triggered on missing MQTT broker

aderusha, 2018-04-25:
The WiFi autoconfig process is triggered when the system cannot connect to WiFi with the saved SSID and password.  This leaves the system in a loop trying to connect to an MQTT broker that hasn't been defined.  The OTA service will work while in this loop, allowing you to push new firmware to the device while it's attempting to connect to MQTT, but that's not a great approach.  I think it might be better to trigger the WiFi autoconfig process if there is no saved MQTT broker defined.

zombu2, 2018-04-25:
great idea

aderusha, 2018-04-26:
With release v0.2.1 I've added a web admin panel which removes the need for this.  If the device can connect to WiFi, it opens an admin interface for basic configuration, allowing you to change MQTT settings and displaying current MQTT connection status along with any errors.  With WiFi up and running, you can now make any other configuration changes as needed.

closed 2018-04-26 by aderusha