aderusha/HASwitchPlate #16: Can't Open Sketchup File

jeffthomps, 2018-06-09:
I have SketchUp Make 17 which as far as I can tell is the latest version of the downloadable free version.
When I try to open the current HASwitchPlate.skp file it says I need version 18. I can download the trial Pro version of 18, but will only have 30 days to work with it. Can you save the file be saved as version 17?
Are you using SketchUp 18 Pro?
Jeff

aderusha, 2018-06-10:
That's a damn good point you've raised.  I am in fact using SUPro 2018, but I should be saving as 2017 so Make users can work with it.  I've done that and thanks for pointing this out!

closed 2018-06-10 by aderusha

jshank, 2018-12-31:
Latest version is saved in SUPro 2018 again. Can't open with Make anymore.