aderusha/HASwitchPlate #20: Change MQTT state topic names

rohankapoorcom, 2018-08-25:
Is it possible to break up the page and box number portion of the mqtt topic string?

from: `hasp/plate01/state/p[1].b[1]`
to: `hasp/plate01/state/p[1]/b[1]`

This would allow simplifying several of the automatons in home assistant so that they can do something like this rather than specifying each trigger manually per page
```
- alias: hasp_down_plate_p0_PageButton1Action
    trigger:
    - platform: mqtt
      topic: 'hasp/down_plate/state/+/.b[1]'
      payload: 'ON'
    action:
    - service: input_number.set_value
      data_template:
        entity_id: 'input_number.hasp_down_plate_activepage'
        value: '{{states.input_number.hasp_down_plate_pagebutton1page.state|int}}'
```

rohankapoorcom, 2018-08-25:
This looks like it should be fairly simple to change in the arduino code. Any objects @aderusha if I give this a shot?

aderusha, 2018-09-08:
I have serious reservations about this, as it runs against a general principle at work here.  My goal is that the Arduino code should be, as much as possible, a simple gateway between MQTT and the Nextion instruction set.  I certainly recognize the advantage for subscriptions when splitting apart the Nextion component naming, but doing so runs counter to that basic principle as you're now splitting up Nextion commands to be parsed by the Arduino sketch.