aderusha/HASwitchPlate #24: Custom Component?

rohankapoorcom, 2018-09-08:
We might be approaching the point where a custom component may be the right way to go rather than having all of the automations dealing with the MQTT directly.

Pull requests #17 , #19, #23 have been moving us towards the right direction (using python_scripts) for the actions being performed but it would be better if we created a new event that would be fired whenever a button was clicked with the button information in the event data. This would simplify the automation triggers.

Additionally, we could fire events when a HASP unit was connected/disconnected (similar to my [custom automations in my own config](https://github.com/rohankapoorcom/homeassistant-config/blob/master/packages/hasp/plates/plates_components.yaml#L78)

Another benefit of the custom component approach is that some point after making a custom component we could work to get it added to Home Assistant proper which would make setting up new HASP units much easier.

aderusha, 2018-09-08:
I don't really speak Python so a custom component is out of my reach.  However, if you wanted to take a swing at it I'm 100% behind the concept.  I'm going to have a difficult time contributing anything meaningful towards the component but would be thrilled if someone were to take it on!

rohankapoorcom, 2018-09-09:
Sounds good, I've made some progress towards this today. If anyone wants to try it out, it's currently on [my fork](https://github.com/rohankapoorcom/HASwitchPlate/blob/custom-component/Home_Assistant/custom_components/ha_switchplate.py)

So far, this handles all of the same stuff as the python_scripts I had added in pull requests #19, #23 as well as firing an event when a node action occurs.

Feel free to assign this issue to me. I'll keep posting here when I make meaningful updates.

aderusha, 2019-01-03:
Coming back to this - I'm still on the fence about a custom component, as it creates another environment for HASP users to learn if they want to make changes to anything.  I've tested the component and it works great!  However, we now have the ability to send a JSON object which allows Home Assistant to publish a single MQTT message with up to 4kb worth of Nextion commands which has had the impact of improving performance and reducing network traffic.  This makes me wonder if the custom component is still a net benefit.

What I'd like to do is to merge the component itself so people can experiment with it on their own if needed, but leave the stock automations as they currently exist (meaning, using the core MQTT functionality) as I believe it will be a little more approachable for new users.  

Does this sound reasonable?  I really appreciate your work here and sorry this has taken so long.  I don't want to sideline the massive effort you've put into this and I'd like to make sure the work you've done is available for everyone.

rohankapoorcom, 2019-01-26:
I haven't seen this JSON object yet. Docs link?

I think the custom component is still helpful with batching these commands. For example, you can provide it a set of key value pairs representing text to update at once and the component can add the necessary logic for adjusting font sizes and batching that all at once so the user doesn't have to worry about it. I think it's also super convenient to just get the sensors provided with no configuration.

I'd like to refactor the custom component (over the last few months, I've learned a lot more about how HA code is architected) and I think it's worth basically rewriting the component. I think it's probably something we can get included in stock Home Assistant rather than as a custom component as well. That makes it much more accessible immediately.

I definitely think the component is a much better approach than the python_scripts in #19.

aderusha, 2019-06-11:
Hey @rohankapoorcom is this still under development?  Looking to cleanup issues and branches here.  Does the JSON command approach outline above do what you would expect?

closed 2020-02-13 by aderusha

rohankapoorcom, 2020-02-13:
The JSON command does do what I expect, but I'm not sure I like the way a full component makes this work super well. It gets a little inflexible in some places.

Let's keep it closed.