aderusha/HASwitchPlate #26: Questions on initial config

rlust, 2018-10-19:
This is a great project!

Just getting started with the initial demo.

How do I set the time zone?

aderusha, 2018-10-19:
If you're seeing the clock displayed on the screen, that's coming from [the example clock automation](https://github.com/aderusha/HASwitchPlate/blob/master/Home_Assistant/packages/plate01/hasp_plate01_p2_clock.yaml).  This automation tells Home Assistant to send the current time every minute, which the panel receives and then displays.

All of this is to say that Home Assistant is the system keeping the clock (and, more generally, is the system which controls nearly every detail of what you see on the HASP display).  To set the time zone on Home Assistant, check [the documentation here](https://www.home-assistant.io/docs/configuration/basic/).

rlust, 2018-10-19:
Thanks, that was a dumb mistake, go it now.
Where do I configure the lights that are controlled?
I assume I can set some of my existing light, but not sure how to configure.
BTW this is really cool! What I have been looking for.

aderusha, 2018-10-19:
Everything is handled in the Home Assistant automations.  The provided automations are there to use as an example and starting point, but to really make this work in your environment you're going to have to build your own automations to suit your own use cases.

You'll find the page 1 scene control button examples [here](https://github.com/aderusha/HASwitchPlate/blob/master/Home_Assistant/packages/plate01/hasp_plate01_p1_scenes.yaml), the page 3 toggle buttons [here](https://github.com/aderusha/HASwitchPlate/blob/master/Home_Assistant/packages/plate01/hasp_plate01_p3_toggles.yaml), and the page 4 dimmer sliders [here](https://github.com/aderusha/HASwitchPlate/blob/master/Home_Assistant/packages/plate01/hasp_plate01_p4_dimmers.yaml).