aderusha/HASwitchPlate #30: Relay control?

jshank, 2018-11-28:
Could this setup support a single relay for local control of a device?

jshank, 2018-11-28:
Could also use this to add dimming on the PWM channel of the WeMos - https://www.aliexpress.com/item/AC-Light-Dimmer-Module-for-PWM-control-1-Channel-3-3V-5V-logic-AC-50-60hz/32802025086.html

[image]

jshank, 2018-11-29:
Thanks for getting back to me @aderusha. The HASP seems to be the best solution available today for dynamic control. I've ordered all of the parts for 2 of these units in an effort to replace some failing [Insteon Keypadlinc](https://amzn.to/2TTvkSe) units. I've looked for comparable Z-Wave solutions and come up very short. I've yet to find a single-gang solution that locally controls a dimming load as well as provides dimming control of 7 other devices. 

I'm really excited at the possibility of context changing the buttons but bummed that I'm losing the ability to locally control the one device like the Keypadlinc does. It might be a while before the displays are here (AliExpress), can you provide the open area inside the enclosure where I might be able to add a relay/dimmer?

aderusha, 2018-11-29:
I think your best bet would be to grab something like a Shelly 1, which should fit in your box with the HASP.  If the ask is to modify the enclosure to provide more room, I'm probably not going to be adding that to the project directly although it should be doable for your own use with the provided models.

jshank, 2018-11-29:
Just wondering if there was room to incorporate the dimmer/relay into the HASP enclosure directly. The Wemos has additional GPIO capacity and I'm sure the sketch could be modified easily to control it. Unfortunately neither Shelly nor Sonoff makes an AC dimmer module.

aderusha, 2018-11-29:
The WeMos has a bit of room above it for mounting of WeMos shields. [There is approximately 12mm of usable space between the WeMos and the LCD above](https://i.imgur.com/Fv12siN.png).  If you need more room than that (and you certainly will for AC dimmers that can handle several hundred watts of load), you'd probably need to modify the enclosure.  The source files for the enclosure are available and editable via [SketchUp Make](https://www.sketchup.com/download/make).

xHirscHx, 2021-02-25:
I know this an old post but was wondering if the new .41 is going to have gpio support.  I want to use a Sonoff T3 as the base for power and the relays and retrofit the mini and screen.