aderusha/HASwitchPlate #31: Faceplate Design Methods

jshank, 2018-12-03:
I'm working on recreating the faceplate in a parametric CAD tool (Fusion 360) for easier customization. I was wondering how you came up with the placement of the panel in the faceplate. I noticed the top of the screen is 30.34mm from the top of the panel (in 3D_Printable_Models/HASwitchPlate_front_double_decora_right.stl) and 35.16mm from the bottom edge. Was there a specific reason it's off center?

jshank, 2018-12-03:
[image]

aderusha, 2018-12-04:
That's because the actual display device (including the frame around the LCD display area) is not centered.  There is very little room between the screw holes for everything to fit, so the dimensions as you are seeing in the model are a requirement to make this work in a North American workbox.

jshank, 2018-12-04:
Thanks. I'll model mine after yours then.

closed 2018-12-04 by jshank