aderusha/HASwitchPlate #33: Too many MQTT messages.

marciogranzotto, 2018-12-09:
The idea of the project is great.
However, it's very message intensive and that demands a lot of the broker. It's causing my broker to slow down for the rest of my devices.

I propose having a batch topic, that we could send one message containing several topics and values, much like the [Tasmota Backlog command](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#using-backlog).
A simple JSON would suffice, like so:
```json
{
	"command": "dim=50",
	"command/page": 1,
	"command/p[1].b[4].txt": "Lamp On"
}
```

aderusha, 2018-12-09:
This is an interesting idea and I'll look into it further.  The latest release of the Arduino code + Home Assistant automations may resolve a lot of the MQTT traffic you're seeing, so I'd make sure you're running the latest automations and that will likely address your current problems.  

Still, I think sending a JSON over with a set of commands it interesting and warrants further investigation.  Thanks for the suggestion!

aderusha, 2019-01-02:
This has been added in version 0.33 using a JSON array instead of objects, as key:value doesn't really make sense here.  Documentation [here](https://github.com/aderusha/HASwitchPlate/blob/master/Documentation/06_MQTT_Control.md#command-syntax).

closed 2019-01-02 by aderusha