aderusha/HASwitchPlate #39: information for pcb

lscosta90br, 2019-01-03:
What are the dimensions height x width mm?
How many layers and thickness?

aderusha, 2019-01-03:
60mm x 40mm, dual layer, silkscreen on both sides, FR4 1.6mm board, 1oz copper, and the tracks/drills/etc are all big enough and far enough apart that they should fit into any board shops' design rules.

MrClowny666, 2019-01-04:
I just wanted to say thank you for this amazing project, it was something I was going to start looking at and then suddenly via DrZ channel I found out that you've already done all the heavy lifting!

I've never completed any PCB's before, so thank you once again. There are a lot of options on the AllPCB page - can I just leave these as standard and use the options you have listed above?

aderusha, 2019-01-04:
Yup!  I personally like how the white solder mask looks so I'd check that box but it's just an aesthetic thing :D

closed 2019-01-10 by aderusha