aderusha/HASwitchPlate #41: Is anyone working on a UK version

nodecentral, 2019-01-14:
As the UK have a different style of wall light switch, I was just wondering if anyone was working on something that would be compatible there ?

aderusha, 2019-01-14:
If I can get somebody to mail me a UK standard workbox I can work on developing a model to fit.  I've been unable to source them locally.

aderusha, 2019-01-14:
What I don't see are detailed IDs of the opening in the wall (the workbox itself).  There is already a [UK model](https://github.com/aderusha/HASwitchPlate/blob/master/3D_Printable_Models/HASwitchPlate_front_single_UK-UNTESTED.stl) based on a similar diagram but I have no way to test if it works, and I haven't gotten anyone in the UK to try printing it for me and reporting back.  If you have a printer, maybe give it a go and let me know what comes of it?

I'd really like to offer some international versions, but to do that I need detailed mechanical drawings (not just screw hole locations, but the entire model), and I'd strongly prefer having something physically in hand so I can mount it and do thorough print/fit testing.

welbymcroberts, 2019-02-10:
There are also a few other points of note for UK folks.

We have 3 back box depths
- 16mm - not common
- 25mm - most common
- 35mm - not common

A number of new build homes have started using the 16mm, which makes it exceedingly challenging to work with if you've got a 3 way - 3 wire switch with 1.5mm cable, however i would argue its safest to target the 25mm, as that's what most folks have.

The biggest issue however is the wiring in most boxes. Most homes in the UK have a '2 wire' light switch, which is a 'switched live', relying on the load to be connected for the switch to have a connection to neutral via the load. A '3 wire' system I believe is similar to the US where Live and Neutral are in the switch box, however we don't actually use the neutral for anything, other than putting on a connector block. Due to this we have a very limited number of relays etc that work with home automation systems, with the majority being things like LightwaveRF.

There also is the added challenge of light switches that are multiple individual switches. It's extremely common to see a 1, 2 or 3 gang light socket, all fit into the same '1 gang' socket shown above (I've not seen anything larger than a  3 switch in a 1 gang, as I think that would be impossible to fit between the two screw holes!). Which can mean even if you're not switching loads with the switch, you've got a plethora of wires to contend with.

I'm looking at options to adapt the existing PCB design into an 'u-shape' pcb that could be used in an existing light switch, without the screen. Mainly as our options for switches that don't actually switch the load directly are very few and far between, are usually touch button, or momentary buttons, both of which are not in line with what most people 'expect' for a light switch in the UK. I know its a different use case to the HASP, but it's something that I'm looking at.

I'd be happy to send some backboxes, and a switch or two for you to use for sizing.

aderusha, 2019-06-11:
Duplicate of #46

closed 2019-06-11 by aderusha