aderusha/HASwitchPlate #42: LCD Reset

RDickens, 2019-01-14:
The `nextionReset()` function does not work.

You have set the `digitalWrite(nextionResetPin, LOW);` but did not set it HIGH again

`void nextionReset()`
`{`
`  digitalWrite(nextionResetPin, LOW);`
`  Serial1.print("rest");`
`  Serial1.write(nextionSuffix, sizeof(nextionSuffix));`
`  Serial1.flush();`
`  delay(2000);`
`  digitalWrite(nextionResetPin, HIGH);`
`}`

aderusha, 2019-01-17:
Great catch!  I had intentionally done this the way it's written as the use cases I was considering were followed by an `ESP.reset()`.  Of course, I forgot the MQTT command `lcdreset` which *doesn't* also reset the ESP.  

Next release will have this fix, thank you for digging into it, pointing it out, and providing an easy solution!

aderusha, 2019-02-07:
New reset code in latest HASP release v0.35

closed 2019-02-07 by aderusha