aderusha/HASwitchPlate #43: TJC display

Supersimo88, 2019-01-17:
I bought what the seller claimed to be a nextion display, but I only discovered after receiving it that it is a tjc instead, can I do something? there is the possibility of having a tft file for this too

aderusha, 2019-01-17:
You will find TJC TFTs and HMIs in the repo [here](https://github.com/aderusha/HASwitchPlate/tree/master/Nextion_HMI).  Support for these devices is limited - online LCD firmware upgrades won't work, and editing the HMI is possible... but the tool to do so is in Chinese.

Outside of that everything else should work using the rest of the code provided here.

Supersimo88, 2019-01-17:
I'm sorry, I do not know why but I had directly copied the file for nextion, not being aware of this difference, on a folder in my computer and so I had not noticed that here was also the version for tjc

aderusha, 2019-01-17:
No worries!  This has happened to a few people which is why I sat and re-created the entire thing in a Chinese language editor (ugh).

Supersimo88, 2019-01-17:
you're awesome

closed 2019-01-24 by aderusha