aderusha/HASwitchPlate #46: EU version (work in progress)

zdar, 2019-01-21:
Hi,
I'm trying to develop a version which fits into EU style box (68mm diameter). 

For those interested  - current version of PCB can be found here: https://easyeda.com/zdar/d1-power-board

At this moment I have prototype PCB assembled and I'm waiting for the LCD to test everything. 
Unfortunatelly the HMI LCD is a bit too big so regular switch front frame and plate will need adjustment (Logus90 in my case).

The design uses power source (the break out part of PCB - simply snap it off by pliers) perpendicullary attached to the main board via solder bridge or angle pins. The LCD should go on top and hopefully it will fit into regular plastic frame with the inner fence trimmed off in apropriate parts.

Comments are welcome :-)

zdar, 2019-01-22:
Open the project at EasyEda and select "Switch version"
Note: It is completly untested with IRM-15-5 and it may be too deep for your box.

stijnbrattinga, 2019-02-17:
Did you already get the Nextion screens? Is everything fitting well into the wall box?  I live in The Netherlands and would like to start a build. Can I use your PCB design as is and just download the .gerber and send to JLCPCB, or do you recommend any additional modifications?

zdar, 2019-02-18:
> Did you already get the Nextion screens? Is everything fitting well into the wall box? I live in The Netherlands and would like to start a build. Can I use your PCB design as is and just download the .gerber and send to JLCPCB, or do you recommend any additional modifications?

I'm still waiting for my 2.4" display. Sofar I have assembled PCB and it fits in deep box (depth of 35 mm is needed when powerboard is attached). 

[image]
[image]
[image]

stijnbrattinga, 2019-02-21:
Very nice! Looks great, I will order some PCB's!

zdar, 2019-02-21:
> Very nice! Looks great, I will order some PCB's!

Wait a few days :-)
I should pick up my nextion shortly (it is at post office) and I'll test everything togeather...

aderusha, 2019-02-22:
WOW!

stijnbrattinga, 2019-02-23:
Very cool, thanks for all the hard work @zdar ! I will order the parts and report back on my progress.

stijnbrattinga, 2019-03-22:
@zdar  Did you use the exact same parts for the rest of the projects? Like the 4pin header and grommet?

zdar, 2019-03-23:
> @zdar Did you use the exact same parts for the rest of the projects? Like the 4pin header and grommet?

No. My LCD is solded directly to PCB by short uninsulated wires. The LCD sits directly on PCB and solder pads on LCD&PCB should be aligned. 

Originally I planned to use plastic spacers and nuts to hold LCD but at the end I just removed connector from LCD board, soldered wires on LCD board in place of connector, aligned LCD on PCB with wires going trough, soldered wires to PCB, did some testing and hot glued the LCD to PCB. That way I got lowest profile.

For the power lines make sure to use good 230V or better rated connector on power PCB. Insulate any exposed mains connected parts (for your own safety). If you want to be extra safe use shrink wrap/insulation tape over power board (do not over do it - the power supply needs cooling).

It all fits into wall box which itself provides secondary (mechanical) protection. Keep it mind that it has to be installed according to your local code and it is intended for permanent installation (not to be opened by user).

If you do not install it into the wall then you need a compliant box. These boxes shall come with proper grommit/cable gland a there has to be strain relief on the cable to prevent cable beeing pulled out and wires exposed.

Warning - this project is connected to the mains. It should be only done by person with proper qualification. There is serious risk of injury (the same applies to orginial US version).

marrobHD, 2019-06-14:
Were can I buy it?

Misiu, 2019-11-18:
@zdar did you use nextion LCD? or maybe other types could be used?

aderusha, 2019-11-18:
> @zdar did you use nextion LCD? or maybe other types could be used?

The HASP project fundamentally relies on the Nextion command protocol, so unless another LCD panel also supports that protocol, you're stuck with Nextion or TJC.  [There have been reports of people swapping out the display panel on an existing Nextion board](https://unofficialnextion.com/t/9-500nit-screen-on-7-nextion-enhanced-board/331) to utilize larger or brighter displays.

rexkani, 2020-04-02:
Hi, may i know why is this L shaped designed needed for EU boxes? i guess the original design give a slimmer profile for the whole thing? with only the thickness of the IRM-03-05 module? 

but this L shape will make the whole thing digging deeper into the box with the length of the IRM-03-05 module?

I'm going to try installing HAswitchplate to our UK boxes and i'm also worried about fitting the whole thing into the box, appreciate if you could share more experiences here.

Thanks!

zdar, 2020-04-02:
> Hi, may i know why is this L shaped designed needed for EU boxes? i guess the original design give a slimmer profile for the whole thing? with only the thickness of the IRM-03-05 module?
> but this L shape will make the whole thing digging deeper into the box with the length of the IRM-03-05 module?

It is not exactly mandatory :-) But there were several reasons for this design. The biggest constrain was simply that there is not enough space in round 68 mm diameter box used on continent (screw distance 60 mm). It is impossible to fit D1 mini and power supply safely there on one board. That means a second board or separate supply had to be used.

I have opted for L shape design for two reasons:
- I did not want to have connecting wires and two separate units in the box
- It reduced depth of assembly a bit.

This construction resulted in 35 mm of depth needed  which fits into commonly used deeper variant of installation wall boxes.

Anyway, you do not need to use L shape construction. You can simply keep the power board separated (and have it parallel to main board or elsewhere) or not use it at all.

I believe that UK you have square boxes which gives you more space. You can probably stick IRM-03-05 somewhere to the corner without any problem which will reduce overall height.

My recommendation - if you have the option install as deep wall box as possible (UK  ?47 mm deep). That way you will have some space for the wiring,...

Note: I do not recommend having power supply on the front PCB (or similary exposed) as it poses risk having live mains lines in place where somebody could touch them after removal of front plate.

rexkani, 2020-04-08:
Thanks for you information, i guess i could try with the original PCB
first, and if it doesnt fit to the UK box, then i will try on your design.

As i'm living in an apartment and i have asked workers to drill deeper
space for boxes during renovation, i guess it should be easier for me.

socialstijn, 2020-07-21:
My soldering connection (Power to PCB and Screen to PCB keep breaking. Any tips on making a small and strong connection from screen to the pcb?

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha