aderusha/HASwitchPlate #49: HA Invalid config

xenio, 2019-01-27:
I am using Hass.io and after the command from cli, the package was created but now I am getting an error on the configuration.yml

Invalid config for [automation]: Invalid platform specified @ data['trigger'][0]. Got None. (See /config/configuration.yaml, line 807)

Any help?

aderusha, 2019-01-27:
That sadly is the result of [breaking changes introduced in Home Assistant version 0.86.](https://www.home-assistant.io/blog/2019/01/23/release-86/#noteworthy-breaking-changes).  Specifically, the `time` component which we use in several automations has been renamed to `time_pattern`. 

You can either upgrade Home Assistant, or run `deployhasp-pre086.sh` instead of `deployhasp.sh` to setup your existing installation.  Note that using the earlier version will break once you upgrade past 0.86.

If this is annoying to you (it's hugely annoying to me), I recommend voicing your thoughts on the thread [here](https://community.home-assistant.io/t/0-86-new-lovelace-ui-and-zigbee-management-panel/93897/89).

xenio, 2019-01-27:
Ok thanks, I just fixed the time/weather tracking back the changes in the latest commit.
I'll check my settings and I'll update.
Thanks.

aderusha, 2019-02-07:
Updated automations available in latest HASP release

closed 2019-02-07 by aderusha