aderusha/HASwitchPlate #51: Temperature sensor

spiddder, 2019-02-08:
Are you planning to integrate a DHT22  sensor into your firmware?

aderusha, 2019-02-10:
HASP is intended to be a control surface as opposed to an all-in-one hub, and I don't know how having a temperature sensor mounted inside the device is going to tell anyone anything useful.  I'd think you'd be far better served hooking a temperature sensor to a device that's not inside of your wall (and thus exposed to the room temperature), and then sending that data out via MQTT, where you can then display that info on the HASP.  Does that make sense?  Am I misunderstanding the question or your use case?

spiddder, 2019-02-11:
I thought it could be used as a thermostat if there was a temperature sensor in it. There was only one idea about me. Maybe someone else would like it.
[image]

aderusha, 2019-02-11:
That actually makes a bit of sense.  Let me stew on this a bit, thanks for the idea!

aderusha, 2019-06-11:
Still think is is really cool but also out of scope of the HASP project itself. Closing issue.

closed 2019-06-11 by aderusha