aderusha/HASwitchPlate #52: Add JSON "status" topic for commands from HASP

bkpsu, 2019-02-10:
With the current implementation, the JSON "command" MQTT topic is great for combining multiple HA automations/openHAB items into single JSON string. However, for the return path (HASP button/dimmer events), dedicated MQTT topics are still required for each button or dimmer value. I think it would make sense to create a JSON "status" topic, that (much like the command topic) would send all HASP events onto a single MQTT channel. The home automation bus could then parse out those events to figure out which button on which page was pressed, and act accordingly. This could be as simple as sending the current MQTT topic id with the associated state, i.e.:

{ 
"event":"p[2].b[1]",
"value":"ON"
 }

when button 1 on page 2 is pressed, or

{ 
"event":"p[4].b[7].val",
"value":"120" 
}

when dimmer 7 on page 4 is moved (preferably on release, so only a single message is sent).

This would greatly reduce the number of MQTT topics associated with each HASP.

aderusha, 2019-02-10:
So here's the challenge with this - to do this, we would have to start queing up commands as they are received from the Nextion.  Nextion only sends out one response at a time, and when we receive that response we send it out immediately.  If we want to batch it up, now we need to setup a queue, make some decision when the queue is ready to send, then send it all at once.  I'm trying to imagine a situation where this would result in anything except higher latency overall, particularly as it's traffic being sent from the panel which is substantially less than traffic being sent to the panel for most users.

What problem are we solving with this approach?

bkpsu, 2019-02-10:
Ok, so I may have jumped the gun and done it already (https://github.com/bkpsu/HASwitchPlate/commit/fd885d2747a18d67b3e995c6b4c0a1ebdbcefd65) :)

It only sends the event to the JSON topic at the same time it's handling the subtopic publishing, but that should work just fine for my application.

Basically, the MQTT binding in openHAB, requires explicit definition of each topic that's going to be read/written to. For all the possible buttons/dimmers on every page of the HASP, that means I have to create an Item (state object in openHAB) for every single button. If I get the event data (button press, dimmer value) as a payload on a single topic, I can set up a single item, and handle the user interaction within a rule.

I don't know if this applies to Home Assistant (or if it could help simplify the automations in any way), but it certainly does not seem to affect the code behavior in any way (other than a second publish for each event).

PR #54 (2019-02-18): JSON state topic for button ON states and dimmer values and greatly reduced OH config
-Fixes #52 
-Adds one state/json topic with JSON payloads corresponding to button ON events and dimmer .val changes.
-Removes *all* explicit item definitions for every single page/button in openHAB configuration, keeping only a single MQTT item to receive the JSON status events (cut down items file by ~250 lines)
-Modifies openHAB rules to handle the JSON states (cuts down rules file by ~100 lines)

* - I'm creating a PR, since this affects the Arduino code and not just the OH configuration. The impact should [... cut]
commit fd885d27 (2019-02-18): Added /state/JSON MQTT subtopic, which publishes all button and dimmer events, with values (ON/OFF for buttons, ON/val for dimmers) in JSON format. Resolves #52 in upstream
closed 2019-02-23 by aderusha