aderusha/HASwitchPlate #58: Bluetooth presence detection

penright, 2019-03-11:
Feature Request:

[This link is an example of someone using esp32 and mqtt.](https://community.home-assistant.io/t/esp32-based-ble-tracking-for-mqtt-room/71309)

[This link is an example of someone using a pi-zero, this has some algorithms that might come in handy.](https://community.home-assistant.io/t/presence-detection-ble-mi-flora-rf-gateway-with-esp32-openmqttgateway-is-now-compatible/37577)

I thought I would start this here so it does not get lost in the main thread, and I am interested in helping code it.  
I have a dozen esp8266 but no esp32 yet. I have one ordered, once I get it, I will try and get a hello world running. Then we can see more what is involved.

aderusha, 2019-03-25:
I think the major challenge is going to be making an ESP32 board fit into the existing WEMOS D1 Mini footprint.  There is a D1 mini-like ESP32 which is pin compatible but it's too wide for the PCB.

penright, 2019-03-25:
When I posted the FR I did not notice the size difference of the ESP8286 mini vs the ESP32.
I see what you mean. I do have the board I order from you now. I also had a B114R (old work box) laying around, and the current .STL files will not fit in it, little on making it bigger. My house has all new workboxes, so I am sure it will fit them.
I do have one of the ESP32 on order and I am waiting on the slow boat.  Still learning. (a lot). :-)

To be honest, I am not sure if it will help or not. Right now for presence detection, I am using the google maps. For some reason, in the middle of the night, it will drift far enough that it triggers a "not home". I try adding nmap, the problem is leaving takes too long to trigger the "leave home". Comparing nmap to google map, arriving is pretty close, leaving is sometimes up to 10 minutes. I know there are settings, but I am trying to save batteries.

aderusha, 2019-06-11:
Closing issue as it pertains to ESP32, not used in this project.

closed 2019-06-11 by aderusha