aderusha/HASwitchPlate #59: MQTTClient

abzaatari, 2019-03-21:
Hi, I download all library (MQTT , wifimanager, arduinojson) and when I compiling the HASwithPlate code in Arduino IDE "MQTTClient does not  name a type" is reported. 

Please any help from you I will appreciate.

thanx

aderusha, 2019-03-22:
Can you give me versions on the setup you have running?  Which library versions specifically, along with the ESP8266 board manager version?

abzaatari, 2019-03-23:
Hi, 
-This is title of the ESP8266 board version " esp8266 by ESP8266 Community version 2.5.0 INSTALLED"   after the update  I restart arduino IDE not my PC . 
-WiFiManager by Tzapu Version 0.13.0 INSTALLED.

- MQTT by Joel Gaehwiler Version 2.3.4 INSTALLED
   Thanx

abzaatari, 2019-03-23:
Hi. Thanks for you help, it works after including   <MQTTClient.h>  in sketch and it should be befor MQTT.h otherwise it will not work.

aderusha, 2019-03-25:
I'm still not certain what is going on here.  I've just downloaded a fresh install of Arduino 1.8.9 (Windows), and deployed the board manager and libraries [as documented here](https://github.com/aderusha/HASwitchPlate/blob/master/Documentation/01_Arduino_Sketch.md#arduino-ide), and the project compiles with no warning and no changes required.

closed 2019-06-11 by aderusha