aderusha/HASwitchPlate #60: "No matching function for call to HTTPClient::begin(WiFiClient&, String&)"

petergreis, 2019-03-25:
Arduino IDE version 1.8.9 on MacOS

Could we get a list of the required libraries (and versions)? I am getting various compiler errors (like the one above), and also a complaint about MDNS.

Second question - can the binary be downloaded to the board directly from the IDE? I don't see a way to do this...

aderusha, 2019-03-25:
Good call.  I've updated [the documentation page](https://github.com/aderusha/HASwitchPlate/blob/master/Documentation/01_Arduino_Sketch.md#arduino-ide) to call out specific versions of the board manager and libraries in use.  I've just tested this in a clean 1.8.9 using the instructions provided and the project compiles without errors.

The documentation also covers using NodeMCU Flasher for writing the binary directly but that's a Windows-only tool.  I don't have a Mac with which to test MacOS binaries but a google search suggests there are several options available.  [This command line tool from Espressif](https://github.com/espressif/esptool) seems like a good starting point.

closed 2019-03-25 by aderusha

petergreis, 2019-03-25:
Hi Allen

OK, flashing with esptool.py does the trick. But after updating my WiFi and MQTT settings, the screen simply displays

WiFi Connected: 192.168.1.205
MQTT Connected: 192.168.1.201

How does it move to the next screen? Is there something that I need to do?

Best regards

Peter

aderusha, 2019-03-25:
You'll then need to proceed through the rest of the documentation to setup Home Assistant.  If you have everything physically hooked up (which is sounds like you do), you can probably jump to [step 5 in the documentation](https://github.com/aderusha/HASwitchPlate/blob/master/Documentation/05_Home_Assistant.md).

petergreis, 2019-03-25:
But what if I am not using home assistant? I use homebridge and an MQTT Broker, eventually driving everything with Apple Home and Siri.

Best regards

P

aderusha, 2019-03-25:
This project includes pre-canned automations for Home Assistant and OpenHAB.  If you aren't going to be using one of those two options, you'll need to build out your own solution for your own environment from scratch.  [The MQTT namespace is outlined here](https://github.com/aderusha/HASwitchPlate/blob/master/Documentation/06_MQTT_Control.md), but understand this is going to be a substantial undertaking.

petergreis, 2019-03-25:
I am ready to do that... was planning on mapping actions through NodeRed. My question is, should it boot fully into the screens with the buttons, or is it waiting to contact a home assistant server?

aderusha, 2019-03-25:
It's waiting on page 0 to be told what to do.  You can tell it to switch pages using the MQTT commands outlined in that document.  If you check the Home Assistant forums you'll find some users have been working on this.  [This user](https://community.home-assistant.io/t/ha-switchplate-diy-lcd-touchscreen-wall-switch-replacement/25464/983) posted his NR flows [here](https://hastebin.com/mayuyegufa.json).

petergreis, 2019-03-27:
Allen

Thank you so much 😊
I now have a 2.4 and 7” version running.

What I did find odd - adding a slider (as p[1]h[0] comes over MQTT as b[2], not h[0]. Is this the expected behavior?

Best regards

Peter

[image1.jpeg]

aderusha, 2019-03-27:
> Allen Thank you so much 😊 I now have a 2.4 and 7” version running. What I did find odd - adding a slider (as p[1]h[0] comes over MQTT as b[2], not h[0]. Is this the expected behavior? 

It is, [check out the documentation here](https://github.com/aderusha/HASwitchPlate/blob/master/Documentation/06_MQTT_Control.md#nextion-page-and-object-ids) for an explanation of what is going on.