aderusha/HASwitchPlate #64: Compile Error - Software Serial - use of deleted function

FritzTheCatfish, 2019-06-02:
This is the Error: 

HASwitchPlate-master\Arduino_Sketch\HASwitchPlate\HASwitchPlate.ino: 

In function 'void debugPrintln(String)':

HASwitchPlate:2506:54: error: use of deleted function 'SoftwareSerial::SoftwareSerial(const SoftwareSerial&)'

     SoftwareSerial debugSerial = SoftwareSerial(17, 1); // 17==nc for RX, 1==TX pin

                                                      ^
Environment is:
Windows 10
Arduino 1.8.9 (Windows Store 1.8.21.0)

Libraries:
Esp8266 2.5.2
ArduinoJson 5.13.5
MQTT 2.4.3
WifiManager 0.14.0

No modifications were made to the file.
Any suggestions?

FritzTheCatfish, 2019-06-02:
BTW what esp8266 version was used? If I change version i.e. 2.5.0 I get an MDNS error

aderusha, 2019-06-11:
Fix is sorted out and coming in 0.38

aderusha, 2019-06-12:
HASP v0.38 has been published which resolves this issue.  HASP now uses the latest version of ArduinoJSON 6.x, and the Arduino build instructions have been updated to match.

closed 2019-06-12 by aderusha