aderusha/HASwitchPlate #73: MQTT Homeseer Integration

mattster98, 2019-07-25:
Trying to plumb the HASP into Homeseer via MQTT.  Most things seem to be working (status, initialization), but for its events, Homeseer seems to require a numeric value (1 or 0) for the button status instead of "OFF' or "ON".

Is that hard-coded, or is there something I can change in either Nextion or the MQTT initialization to change that?  I didn't see it anywhere.

aderusha, 2019-07-25:
That is hard-coded [here](https://github.com/aderusha/HASwitchPlate/blob/master/Arduino_Sketch/HASwitchPlate/HASwitchPlate.ino#L669-L690).

mattster98, 2019-08-30:
Update - my issue was with a plugin, not with what HASP was sending.  No changes needed to ensure compatibility.

closed 2019-08-30 by mattster98