aderusha/HASwitchPlate #75: Fitting into old work electrical box

osbash, 2019-09-07:
Has anyone been able to edit the 3D files to allow the back piece to fit into an old work electrical box?  Something like this:  [https://www.homedepot.com/p/Carlon-1-Gang-Old-Work-PVC-Electrical-Box-with-Clamp-Case-of-25-B114RR-UPC/205338594?mtc=Shopping-B-FBT-G-Multi-NA-Multi-NA-Feed-PLA-Catchall-NA-NA&cm_mmc=Shopping-B-FBT-G-Multi-NA-Multi-NA-Feed-PLA-Catchall-NA-NA-71700000014585962-58700001236285396-92700010802552412&gclid=Cj0KCQjwh8jrBRDQARIsAH7BsXcpqL3JcvexeAqBacrTVfQnoCju0wNXdHsuX5eVDsxm66_w6QXGjs0aAvqJEALw_wcB&gclsrc=aw.ds](old work electrical box)

The inner dimension is just under 2.9"

aderusha, 2019-09-08:
I created a [minimum clearance rear  enclosure](https://github.com/aderusha/HASwitchPlate/blob/master/3D_Printable_Models/HASwitchPlate_rear_lcdmod_minimum_clearance.stl) that you might be able to use.  It is a little flimsy, needs supports to print, and generally isn't as nice as the normal rear enclosure, but it has every bit of non-essential geometry stripped to the bone so it is as small as I could make it for tight work boxes.

osbash, 2019-09-09:
Thanks for the response.  After looking into it a bit more, I noticed the Nextion itself is about 1mm taller than the work box, so it wouldn't work even if I were to reduce enclosure to fit.  I don't have any spots to install the switch plate so I need to add one of these work boxes into existing drywall.

My only other option is to create a 3d printed workbox to work like the one linked above and fix the enclosure.

closed 2019-09-09 by osbash