aderusha/HASwitchPlate #78: Control without MQTT?

m14t, 2019-10-27:
### Question
Is it possible to adjust the backlight level when the MQTT server is unavailable?

### Back story

I live in CA and am effected by the fire preventive power shutoffs.  When this is scheduled, I proactively power down a bunch of things, like the MQTT server that my HASwitchPlate connects to in order to save the UPS it's connected to for other things.

However, when this happens, the HASwitchPlate displays the wifi information and sets the backlight to what looks like 100%.  This plate is in our bedroom and so ideally we would be able to control/dim this backlight.

--

Thanks again for all the work here!  This is one of the first things I look at every morning.  I love it.

petergreis, 2019-10-28:
> Reply to: aderusha/HASwitchPlate <reply@reply.github.com>

In the pre-initialize for my 7” panel I have the following:

thup=1

thsp=120

This blanks the panel after 120 seconds of no input. The first touch thereafter turns the panel back on, the next touch is passed on normally. Perhaps this would help you?

aderusha, 2020-02-13:
The comment above shows one way to deal with the specific nature of the issue raised.  Also, the ask for "control without MQTT" is now fulfilled via telnet.  The telnet client allows you to send commands to the Nextion panel without MQTT being active.

closed 2020-02-13 by aderusha

Thoky81, 2020-08-10:
@petergreis Hi, can you please explain where to put thup=1 thsp=120 in nextion editor? I have done something but the display dont want to sleep - just restart. Plus, when I tried to simulate it on nextion, it was working no issue. Thanks

petergreis, 2020-08-11:
Certainly, it goes in the Preinitialize Event. For example, I have this:

bauds=115200

vis 3,0

thup=1

thsp=120

sendme

best regards

Peter

Date: Monday, 10 August 2020 at 22:17
Cc: Peter Greis <peter.greis@freethinker.com>, Mention <mention@noreply.github.com>

Thoky81, 2020-08-11:
@petergreis Thanks its working now.