aderusha/HASwitchPlate #81: HASP Time not correct since upgrading to HA .104

rlust, 2020-01-16:
The displayed time on all of my panels are running several minutes slow since HA upgrade to V .104.

aderusha, 2020-01-16:
Is the time set correctly on your HA system?  An HA automation is triggered every minute which sends the current time to HASP.  The HASP understands nothing about time, it's just displaying what HA tells it.  If HA has the wrong time, then that's what it sends to the HASP.

rlust, 2020-01-17:
HA time is correct, I was just looking at all of the MQTT hasp messages.
Very few and very slow to update. Everything was fine prior to HA update to .104 and now .104.1.
MQTT server is fine and no issues with other device types using MQTT.

aderusha, 2020-01-17:
So for some reason Home Assistant isn't sending the messages on time, this is going to require troubleshooting your Home Assistant installation to determine why automations using the Time trigger aren't firing on time.

rlust, 2020-01-17:
OK Found the issue. I added more memory to the HA VM. It was maxing out as was the Swap area. I added memory and so far all is well.
Glances was very helpful in finding the issue.
Thanks for the help!

closed 2020-01-17 by aderusha