aderusha/HASwitchPlate #91: Three-gang plate with HASP in the middle

aph3rson, 2020-02-16:
At the moment, there's only one option for a three-gang plate with an HASP module - mounting it on the side. 

In some instances, it may be more desirable to have it in the middle - case in point, I have two ZEN22's I'd be mounting it next to, which can't be mounted next to each other (heat dissipation issues). Therefore, putting the HASP in the middle would be more practical and fits the aesthetic I'd be going for.

aph3rson, 2020-02-16:
For reference, here's the two models that I'm referring to: [three-gang Decora](3D_Printable_Models/HASwitchPlate_front_triple_hasp_decora_decora.stl), [three-gang toggle](3D_Printable_Models/HASwitchPlate_front_triple_hasp_toggle_toggle.stl).

aderusha, 2020-02-29:
You make a compelling argument.  [Currently available in dev branch](https://github.com/aderusha/HASwitchPlate/blob/dev/3D_Printable_Models/HASwitchPlate_front_3x_decora_hasp_decora.stl), should hit master "soon".

closed 2020-03-24 by aderusha