aderusha/HASwitchPlate #92: Uploading firmware or TFT to nextion display

jaygrovr, 2020-02-24:
I bought a nextion display from amazon and the part number is NX3224T024_011, and when i try to update the firmware on the device, I get the following message "model does not match Device model: NX3224T024_011R".  What am I doing wrong?  This part number seems to match what your readme file describes,.

aderusha, 2020-02-24:
How are you handling the update?  SD card? Serial through HASP?  Any screenshots of what you're seeing?  Can you provide a link to the device you bought?

jaygrovr, 2020-02-24:
Sorry, doing the update using an SD card.. i formatted it as fat32, and the device recognizes the file, but for some reason doesnt allow it.  Here is the link for the screen https://www.amazon.com/gp/product/B015DMP45K/ref=ppx_yo_dt_b_asin_title_o00_s02?ie=UTF8&psc=1

it does say the exact part number on the PCB.,. without the R at the end..

aderusha, 2020-02-24:
Weird :(  There should only be two models of that device per the Wiki: https://nextion.tech/datasheets/nx3224t024/#2

One ending in "R" for resistive touchscreen, and one ending in "N" for no touch sensor.

Do you have a spare UART kicking around?  I wonder if loading up the Nextion editor and attempting to send the project that way would yield any different results or at least tell you more about what it thinks is happening.

jaygrovr, 2020-02-24:
[image]
[image]

aderusha, 2020-02-24:
Allright one more idea - can you show me the contents of the SD card?

jaygrovr, 2020-02-24:
i can try with a uart, but was hoping to not have to do that.  Could it possibly be not enough power for the screen?
[image]

aderusha, 2020-02-24:
That is the correct file for your display model.  Power problems can present themselves in all sorts of wacky ways but I've not seen it do this...

I'd have to default back to using the Nextion editor.  Once it connects it should identify your device and give you some options for the current project to make sure it matches your LCD.

jaygrovr, 2020-02-24:
ok so with a uart, i should be able to connect to the screen directly with the editor?

aderusha, 2020-02-25:
That's the idea I'm shooting for.  Take HASP out of the middle, let the Nextion software talk directly to the Nextion panel and see what it makes of the situation.

jaygrovr, 2020-02-27:
by the way I figured out my issue.  For some reason github wasnt downloading the correct file.  i was downloading some other file type.  Once I signed in, the files were much larger and now everything works.  Totally my mistake and I feel very dumb about it.
jason

aderusha, 2020-02-28:
Hey brother glad you got it sorted!

closed 2020-03-19 by aderusha

thatoneprogram, 2021-09-19:
how am I supposed to get rid of this error

thatoneprogram, 2021-09-19:
I

thatoneprogram, 2021-09-19:
need

thatoneprogram, 2021-09-19:
help

aderusha, 2021-09-19:
Can you maybe help us out by explaining what you've done, what you expected, and what happened?

thatoneprogram, 2021-09-19:
nvmd i have done it

melazzab, 2021-12-27:
I have the same error but when trying to uplad the tft file via the the serial port using the **whmi-wri** instruction. The same file can be uploaded properly using the SDcard, so the version is normally correct. But when I try to appload it via serial port, I receive the first message back from the screen **(0x5)** meaning that it is ready to receive, then I send the first 4096 bytes of the file, then the screen shows the error : Model does not match

mohammedelazzab, 2021-12-27:
I have the same error but when trying to uplad the tft file via the the serial port using the whmi-wri instruction. The same file can be uploaded properly using the SDcard, so the version is normally correct. But when I try to appload it via serial port, I receive the first message back from the screen (0x5) meaning that it is ready to receive, then I send the first 4096 bytes of the file, then the screen shows the error : Model does not match.

The file is generated using Nextion Editor on windows, then I move it to a Linux machine connected to the nextion touchscreen 3.5" Basic.

aderusha, 2021-12-27:
Which version of the HASPone firmware are you running, and what serial speed to do have set?

melazzab, 2021-12-27:
I don't know what is the HASPone. Can you please explain what it is and how to get the version?
 I am using the Nextion NX4832T035_011R, which is a 3.5" Basic, with a baudrate of 9600, I can establish a communication with the screen and send commands etc with that baudrate, only the TFT upload that doesn't work.

simoelazzab, 2021-12-27:
don't know what is the HASPone. Can you please explain what it is and how to get the version?
I am using the Nextion NX4832T035_011R, which is a 3.5" Basic, with a baudrate of 9600, I can establish a communication with the screen and send commands etc with that baudrate, only the TFT upload that doesn't work.

aderusha, 2021-12-27:
Are you using the HASwitchPlate project?  This repo is no longer maintained and everything has moved over to the [HASPone repo](https://github.com/HASwitchPlate/HASPone).  If you stumbled into this looking for general Nextion help, maybe try the [Unofficial Nextion Forum](https://unofficialnextion.com/).

ageorgiev86, 2023-01-28:
what do we do if the model is correct.  because in this case it is exactly like that.  The model is correct, but it tells me that it does not match

henriquejsbraga, 2025-03-07:
I have the same problem. The model was selected correctly, but the display says it is wrong. My display model is NX3224T024_011R.