aderusha/HASwitchPlate #93: Trouble compiling source

bgardner46, 2020-02-25:
I have an HASwitchPlate running on a breadboard, have printed the case, and am just waiting for PCBs from PCBWay which have been slightly delayed due to the virus problems over there. I am using it with my Homevision controller and an MQTT plugin and it works great. I definitely plan to build several of them. My only problem is that I would like to make some minor changes for my personal use but I have been unable to compile the source. I am using the Arduino IDE version 1.8.8 and the ESP8266 core version 2.5.0. I get numerous errors but they all seem to have this form:

HASwitchPlate:1595:24: error: 'HTTP_HEAD_END' was not declared in this scope

   httpMessage += FPSTR(HTTP_HEAD_END);

D:\Electronics\HASP\Arduino\HASwitchPlate\HASwitchPlate.ino: In function 'void webHandleSaveConfig()':

HASwitchPlate:1716:30: error: 'HTTP_HEAD' was not declared in this scope

   String httpMessage = FPSTR(HTTP_HEAD);

It's probably something simple but I'm stumped. Any suggestions would be greatly appreciated.

Thank you for an incredibly useful project.

Bob Gardner

aderusha, 2020-02-25:
That problem is due to a change in the WiFiManager library.  The next release of HASP will address this issue, but you can clear it up yourself by simply deleting the lines calling HTTP_HEAD_END (so, just delete or comment out the entire line `httpMessage += FPSTR(HTTP_HEAD_END);`).

bgardner46, 2020-02-25:
Well I tried commenting out all of those lines as well as others that were
causing errors but never got the source to compile. I even tried other
versions of the WiFi Manager library but was unsuccessful there as well.
I'm not really in a big hurry so at this point I will wait for your next
release.

Thanks again for releasing such a great project.

Bob

bgardner46, 2020-02-25:
Cancel my last email. I tried one more time with a fresh version of the source and version 0.14 of the WiFiManager and the source compiled. Sorry for the confusion.
Thanks,
Bob

aderusha, 2020-02-28:
No worries.  The Arduino library system can make this difficult, glad you got it worked out.  The next release will be up-to-date vs all current libraries, but that will last for all of a day until something gets updates somewhere....

closed 2020-02-28 by aderusha