aderusha/HASwitchPlate #96: Setting Nextion attributes

bgardner46, 2020-03-10:
Greetings,
I'm not sure this is the correct place to put this so please tell me if there is somewhere else to post.
I am building an HASwitchPlate but using it with my Homevision home control system not Home Assistant. It has been going well but I was having trouble with sending some MQTT commands to set or change attributes on the nextion display from my system. For example I can send:
hasp/disp01/command/page 1
and page 1 is sent to the display and it changes as it should
however when I send:
hasp/disp01/command/picq 0,0,120,120,1
picq=0,0,120,120,1 is sent to the display and I get an error since the display doesn't want the = sign.
I went into the ESP8266 source and changed the following lines as shown here by replacing the = sign with a space.

`void nextionSetAttr(String hmiAttribute, String hmiValue)
{ // Set the value of a Nextion component attribute
  Serial1.print(hmiAttribute);
  Serial1.print(" "); // was "="
  Serial1.print(utf8ascii(hmiValue));
  Serial1.write(nextionSuffix, sizeof(nextionSuffix));
  debugPrintln(String(F("HMI OUT: '")) + hmiAttribute + " " + hmiValue + "'");```

This works for now but I don't know if this will affect anything else or if I'm doing something incorrectly. Can you provide me some insight?
Thanks,
Bob Gardner

aderusha, 2020-03-11:
What you're looking to do is send a command, not set an attribute.  The modification you've made to `nextionSetAttr()` is going to cause problems for you down the road and I'd recommend switching it back, as you'll have broken the ability to send attribute changes.

To run Nextion commands that have arguments, you'll want to send a message like this:

**Topic:** `hasp/disp01/command`
**Message:** `picq 0,0,120,120,1`

aderusha, 2020-03-11:
As a side note - very interested to hear about your progress with getting HASP to work with HomeVision!  If you want to drop in on [our Discord channel](https://discordapp.com/invite/c5DvZ4e) I might be able to field questions interactively as they come up and would love to help out where possible.

bgardner46, 2020-03-11:
Thanks for the quick response. I switched the Wemos software back to the
original form and I changed the way I was sending the message from
Homevision and it worked just fine. From Homevision I send a serial command
which is captured by an MQTT plugin written by one of the Homevision users
which is written in TCL. The original purpose of the plugin was to work
with Sonoff and Shelly devices that had been flashed with Tasmota software.
The command has this format:

mqtt: pub hasp/disp01/command picq 0,0,120,120,1;

This worked great. I also wrote another plugin that captures the published
messages from the hasp and performs the associated actions in Homevision.
When I'm all done I'll put together a page on my web site that has all of
the details and share it with you and our Homevision users.

I will also pop over to the discord channel since I have  few other
questions about 3D printing the enclosures.

Bob

closed 2020-03-12 by aderusha