aderusha/HASwitchPlate #97: group depreciated

mcfarlde, 2020-03-19:
Hassio 107 depreciated groups which is used extensively in the HASP demo configuration.

aderusha, 2020-03-19:
Groups haven't been deprecated entirely, but much of the functionality we had been using has been removed.  The current `master` branch of the project does not include and of the group capability that has been removed and loads on 107 without problem.

If you have an existing installation, a script was created to migrate your existing HASP installations to work with 107.

```bash
bash <(wget -qO- -o /dev/null https://raw.githubusercontent.com/aderusha/HASwitchPlate/master/Home_Assistant/migrate-hasp-107.sh.sh)
```

Finally, the `dev` branch contains the current codebase being developed for better Lovelace compatibility.  [See here](https://community.home-assistant.io/t/ha-switchplate-diy-lcd-touchscreen-wall-switch-replacement/25464/1369) for details on how to test that.

marc-gist, 2020-03-24:
Did something break/change in 0.107.5? as I get a blank page when I click on the "plate01" from the left bar (don't have a "plate01" tab on lovelace).
Thanks!

aderusha, 2020-03-24:
The link to the left is an iframe, which is opening the web interface on your HASP directly and presenting it in a frame.  It relies on MDNS working in your environment to turn "plate01.local" into the IP address of your HASP.  If for whatever reason your desktop can't resolve that address, then that frame won't work.

If you have the IP address of your HASP you should be able to type the address in directly to a web browser and access the same interface.

aderusha, 2020-03-29:
Closed, all compatibility fixes are now in the `master` branch.

closed 2020-03-29 by aderusha