aderusha/HASwitchPlate #101: PlatformIO support

gztproject, 2020-04-08:
Hi, I've been working on creating a PlatformIO version of the ESP firmware, would you be interested in including that in your repo?

aderusha, 2020-04-08:
This is all developed in PlatformIO and the project should be directly portable with no code changes, just drag the .ino into `./src`, pull in your libraries, and it will build.

gztproject, 2020-04-08:
Thanks for quick response.
I had some problems with function declarations but that might be an error on my part renaming it to .cpp

In any case, what I had in mind (and failed to properly explain) is splitting the code in classes as it might simplify it a bit and make the main file more readable. I will probably do it in my fork anyway, and see how it turns out :)

aderusha, 2020-04-08:
Sounds good.  I'm keeping the codebase Arduino IDE-native because I feel like that's the lowest-friction option for new users, but personally I can't stand working in that environment :D

Splitting up the code into classes is something that I should have done from the get-go, and we are now where we are due to years of accumulated features and me being a lousy programmer.  If you want to take a swing at this, please do!

SittingDuc, 2020-12-06:
Hello.

I have forked your project to add some features I want (mostly more pages despite Nextion's limitations) and along the way I am going to try to break the .ino apart into C++ classes and use PlatformIO too because I like the environment better.

Since it is not fair to keep changes to myself, I have tried to make my code less ugly than usual and started uploading it to GitHub.

So far I have a separate class for all the Nextion accesses, and a separate class for all the debugPrintln. I still want to separate MQTT, Webserver and Telnet into more classes.

Since I am walking into your project and tearing it all to pieces on my fork, I thought at the least I should say Hello. You are welcome to have a look at my fork and let me know how you feel. If I have "the right end of the stick", if there are small things I can do differently that would make the class-separating-work more useful to you; if there is anything I can usefully do to make the HASP project better.

Thanks!,

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha