aderusha/HASwitchPlate #106: Include all Open Digital and analog IO to configuration and MQTT protocol

miloit, 2020-05-15:
Hello,
Might it be possible that you can add all open I/O of the Wemos to the mqtt protocol and can they be configured via website to
-input
-output
-dht sensor protocol

So you can individually use the hasp for other purposes as well.

Thanks

PaulG4H, 2020-06-22:
Hello,
if you are searching for an universal Firmware for ESP which is configurable over web and support a lot of devices OOTB maybe ESPEasy or Tasmota is the way you want to go.

kquinsland, 2020-10-17:
@miloit It's relatively easy to add your own code to publish the value of the `A0` pin to MQTT.
[I added a LDR (and neopixels) to the HASP](https://karlquinsland.com/2020/10/enhanced-home-assistant-switch-plate-hasp/) to check the ambient light level in the room where the HASP is deployed.

You can see the code that publishes the LDR value here:

https://github.com/kquinsland/HASwitchPlate/blob/add-neopixels/Arduino_Sketch/HASwitchPlate/HASwitchPlate.ino#L2994

But if you're looking for more 'generic' sensor support, Tasmota and ESPHome are much better alternatives!

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha