aderusha/HASwitchPlate #107: Not remembering wifi after reboot

L337Lobster, 2020-05-17:
Every time the system reboots, be it from loss of power, glitch that forces a reboot, or a command, the QR code comes up and I need to reconnect the wifi. The MQTT broker and admin settings are still there but I need to manually reconnect every time. It also doesn't remember the brightness setting it was at, but that might not be a thing it's expected to do?

I purchased a premade one via your Tindie page if that helps at all.

aderusha, 2020-05-17:
That certainly isn't a thing it's expected to do!  The very first basic step I'd suggest would be to re-flash the firmware through the admin web page, followed by a factory reset.

If that doesn't get you running, can you maybe [stop by the Discord channel](https://discord.gg/tnBsMw4) and we can work through it together?

aderusha, 2020-05-31:
Did this ever get sorted?

aderusha, 2020-12-19:
Closing after no update from submitter

closed 2020-12-19 by aderusha