aderusha/HASwitchPlate #109: no ower issues

xHirscHx, 2020-05-30:
So i built 2 of these and using all the build materials for the list. When i power it with ac the wemo does not power on. If i plug in usb to wemo it powers up but no screen  ( no 5v as expected).
I have 120v to power supply lugs and 5v to the 5v lugs but not to wemo 5v or 3v.

aderusha, 2020-05-30:
Can you post pictures?  Are you using the PCB or protoboard?  In general, I'd advise tracing the path of 5V and see where it stops working.  If you have 5V at the PSU, do you also have 5V at the 5V pin to the WeMos?  If not, can you do a continuity test and see why that is?

xHirscHx, 2020-05-30:
im using the pcb board hasp  from the link. have continuity from 5v power supply to both pins wemo and the pin to the right on pcb.  It seems its a ground issue but i no nothing about this power supply form ac to 12v.

aderusha, 2020-05-30:
Any chance of a soldering issue on the 5V or GND pins on the WeMos?  Can you check your joints on the Wemos pin headers and also on the HASP board? 

The Wemos has a 5V to 3.3V regulator on board.  5V from the USB or from that 5V pin both go to that regulator which then powers the board.  So, if it works from USB, that means that the regulator is working and the board is working.  This leaves getting 5V to that regulator, which must be your problem.

xHirscHx, 2020-05-30:
definitely could be a sodering issues i did my best non are touching but it ant pritty LOL

aderusha, 2020-05-30:
Visually I don't see anything overtly wrong, but the solder joints here aren't great and it's very possible that they aren't making contact where you'd hope they would.  I'd suggest some solder braid + flux to clean things up a bit.  Also, check continuity from +5V on the top of the Wemos board to +5V on the PSU.  Also check GND to GND, and also confirm there is no continuity between +5V and GND.

xHirscHx, 2020-05-30:
No continuity from gnd to gnd. 5v to 5v is fine

xHirscHx, 2020-05-30:
I have ground from psu to the header pins on the right but no ground at the wemo pin

aderusha, 2020-05-30:
That leaves GND at Wemos to it's header pins (on top) or the GND from wemos header pin to the PCB itself on the bottom.  If you have solder flux (not plumbing flux, stuff for electronics), soak those joints and hit it with some heat.

xHirscHx, 2020-05-30:
no joy. will it hurt anything if i conect the  2 gnds with a jumper and leave it

xHirscHx, 2020-05-30:
Funny i have 2 boards that are doing this

aderusha, 2020-05-30:
A jumper from PSU GND to Wemos GND is OK.

xHirscHx, 2020-05-30:
soldered  a jumper No fire no smoke all good booting up.
thx for your help,

closed 2020-05-30 by aderusha