aderusha/HASwitchPlate #110: EU Version for 3.2 inch Nextion with PIR, LUX and BME280 Sensor, Automatic Backlight

PaulG4H, 2020-06-22:
I start with this project and update the software with the following features:

- change the display to an 3.2 inch which will fit in an standard 55mm double frame
- use an BH1750 I2C Lux sensor to calculate the screen brightness.
- the motion sensor switches the display on until timeout without re trigger by the PIR expired
- use an BME280 I2C temperature, humidity and air pressure sensor
- create a 3D printable case with base and top connected by 5x2 magnets

The software works but is not perfect as yours...

[image]

[image]

[image]

[image]

Are there anyone interested?

miloit, 2021-01-11:
> I start with this project and update the software with the following features:
> 
> * change the display to an 3.2 inch which will fit in an standard 55mm double frame
> * use an BH1750 I2C Lux sensor to calculate the screen brightness.
> * the motion sensor switches the display on until timeout without re trigger by the PIR expired
> * use an BME280 I2C temperature, humidity and air pressure sensor
> * create a 3D printable case with base and top connected by 5x2 magnets
> * use it with IP Symcon over mqtt
> 
> The software works but is not perfect as yours...
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> Are there anyone interested?

Hello, I am interested.

Can you post your code and STL files.

SeraphimSerapis, 2021-01-11:
Better late than never. I'd also be interested -- would even consider buying a pre-assembled unit ;-)

PaulG4H, 2021-01-13:
Hello,

sorry for the late reply...

I will fork the project within the next week with my changes and STL Files

PaulG4H, 2021-01-13:
Fork done...

https://github.com/PaulG4H/HASwitchPlate

aderusha, 2021-01-13:
If you want to issue a PR I'd love to roll these into the main project repo!

PaulG4H, 2021-01-15:
> If you want to issue a PR I'd love to roll these into the main project repo!

Hi,

as I see you also still working in the dev branch, so it makes no sense to to do a PR for the code.

My current Version is much ahead and sliced into different files and the development is switched to Visual Studio Code with platformio instead the Arduino IDE (no more library version issues!)

My project is now an universal ESP8266 which I use for controll my 3D Printer Rack (Nextion Enhanced with 8 Relais board connected and controlled by the Nextion), three MQ4 Smoke Sensors on and ADS1015 I2C ADC board, 1-Wire Temperature Sensors and more...

It would be a honor for me if you add the stl files and code parts to your repo, then I will delete my fork again.

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha