aderusha/HASwitchPlate #111: Connecting display using Wemos D1 R1 (not mini)

mirko25, 2020-07-11:
I have Wemos D1 R1 and Nextion NX3224T028_011 panel
I flashed firmware for nextion using sdcard successfully.
Flashed D1 using Arduino - selected WeMos D1 R1 as board also successfully.
Connected blue wire from panel to D7 and yellow to D4.
When powered on I can access web admin and set wifi network and connect to MQTT
But on display there is only message HASP Initializing... 
Do I need to change something in sketch for D1 R1 board or connect display to different pins?

aderusha, 2020-07-11:
Your pinout as described should be correct.  That behavior suggests TX from the WeMos on pin D4 for some reason isn't making it to the RX on the panel.  Confirm your wiring with continuity checks, if you have access to another serial interface you can check at the panel to confirm it's seeing Nextion commands.

mirko25, 2020-07-12:
Thanks for your help. I finally found the problem. tx on my board is on pin labeled d9/tx1 so i switched it and it works.

closed 2020-07-12 by mirko25