aderusha/HASwitchPlate #112: Certificate support?

jshep321, 2020-07-12:
Hi,
I have my MQTT setup for secure TLS / Certificate based usage only.  Does HASP support this?
Thanks!

aderusha, 2020-07-12:
Not as of now, no.  Doing this in a manner which works across all installations and which is also actually secure is problematic as we don't have the room nor the update mechanisms to pull trusted root certs regularly in order to compare your host cert.

It may be possible to do this insecurely, meaning connecting via TLS but not bothering to check the cert.  I'll dig into it.

jshep321, 2020-07-16:
OK thanks.  I'm soon going to be looking into integrating TLS+certs into my ESP32.  Likely similar challenges.   Might move to a REST interface instead of MQTT since TLS is already integrated there.

aderusha, 2020-12-26:
I'm digging into this one with no success thus far.  I think we might be able to do this by offering the user the ability to connect via TLS but with no actual certificate check, or by allowing the user to supply the expected certificate fingerprint.  However, getting BearSSL to work with the RAM currently left available by the rest of the code is problematic.  The most recent commit to the dev branch includes a substantial memory reduction in the form of doing fewer stupid things in the rest of the sketch.  It hasn't cleared up enough RAM to make BearSSL work right, but it's a start.

I've created a `MQTTTLSTEST` directive which turns on the current, broken attempts to make this fly.

aderusha, 2020-12-26:
Found some means to reduce the BearSSL memory requirements and that hasn't impacted functionality with any of the required services (MQTT add-in for Home Assistant and github).  Validation process is still fingerprint-only but it's a start.  `dev` branch should have a fully-working implementation, and it'll likely be pushed out in the next release here soon.

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha