aderusha/HASwitchPlate #114: Track backlight on/off status for Nextion sleep/wake

aderusha, 2020-08-07:
Catch sleep/wake events from the panel and send MQTT backlight ON/OFF status in response.  Great idea @SeLLeRoNe!

jmyler, 2020-08-07:
I love this idea

mattkerrison, 2020-10-05:
Any update on this one? would be handy to wake up the screen with a touch but not accidentally activate a button

aderusha, 2020-12-03:
Also adding - create demo automations which can control sleep/wake state.

cbrochar, 2021-01-09:
I have been using the sleep mode via the automation in examples. I have noticed you cannot change the sleep on touch timer while the screen is asleep. Looing in the instruction set "A get/print/printh/wup/sleep instruction can be executed during sleep mode. "  So if the screen is asleep it ignores the "thup" command. I am able to avoid this by including a "sleep=0" command in the JSON just before the thup command. However, I noticed that you are strating to implement this issue/enhancement. I am wondering how this will effect the backlight on/off since the nextion will not send the 0x87, it should return a 0x68, So you may be able to add the backlight on to the "else if (nextionReturnBuffer[0] == 0x67 || nextionReturnBuffer[0] == 0x68)" statement. This way you do not get off-track when a sleep=0 command is sent.

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha