aderusha/HASwitchPlate #116: {request} - Any chance you can export the sketchup as a STEP file?

kquinsland, 2020-08-18:
First, this project represents a *TON* of work and i'm grateful you've shared it with us! 

I am (attempting) to make a few small modifications to add some functionality. This functionality does require *slight* modifications to the 3d printable enclosures, however.

The CAD package that i am strongest with is Fusion360... which does not have very good support for sketchup file imports. The results are... less than ideal. It appears that the individual polygons / triangle faces are what fusion is able to import. This makes some amount of sense to me as Fusion360 is not a native mesh program... but apparently sketchup is.

I did a quick bit of google and it does appear that sketchup *pro* supports exporting to STEP format, but it's not clear to me if this file / model was developed with pro.

If you can't do the export / conversion then don't worry about it / close this ticket and i'll re-create the model in f360 to make the changes there.

If the changes are successful, i'll publish the STEP files so they should be a painless import into any CAD package.

Thanks again for putting this project together :).

aderusha, 2020-08-18:
I am using Sketchup Pro, but the only means I know of to export STEP from SketchUp is via paid 3rd party plugins.  Annoyingly, I don't know of any particularly good way to bring this into f360.

kquinsland, 2020-08-18:
Ah, you're right! 

I re-read a few of the threads that popped up on my google search and it turns out that you need to use the extension store in sketchup pro to do this. I will try exporting to DXF or just project the mesh and re-do it that way.

Thanks for the quick reply!

closed 2020-08-18 by kquinsland

kquinsland, 2020-10-17:
I was not able to find a clean way to port the sketchup files to STEP, so i re-created them.

After a lot of work, I was able to incorporate the two modifications that I hinted at when I opened this thread:
- NeoPixels /. addressable RGB LEDs beneath the surface of the plate
- LDR / light. sensor beneath the surface of the plate.

Details on the mods: [here](https://karlquinsland.com/2020/10/enhanced-home-assistant-switch-plate-hasp/) and the STEP files are [here](https://github.com/kquinsland/HASwitchPlate/blob/add-neopixels/3D_Printable_Models/enhanced-HASP/combined.step)