aderusha/HASwitchPlate #120: Display works on USB but not thru PSU

exeljb, 2020-08-30:
So I'm trying to get my setup working but I'm having issues while powered up thru AC power.  Everything works fine when I power everything up thru the USB port on the D1 mini.  Then when I plug in the AC (USB power removed) everything powers up, D1 and Nextion but the display doesn't seem to send any serial data nor does it react to it.  I've checked that my PSU is outputting 4.99Vdc and has a ripple of 9mVac, USB power was showing 4.99 and 11mVac so I'm sure my PSU is ok.  Does anyone have any insights on why serial data might stop working when not using the USB power?

xHirscHx, 2020-09-03:
Had same issue  look at my post in closed issues "no power issues"
Check your grounds mine was a ground issue on the board.

aderusha, 2020-09-07:
Nothing specific to offer other than a generic "check your solder joints".  Have you made any more progress with this?

exeljb, 2020-09-08:
I had a chance to mess with it this weekend.  I had tried to clean up and make sure the solder joints were in good shape but the condition persisted.  I opted to just grab another pcb and power supply and put it all together from scratch.  The condition was not present on that new setup.  
I'm unsure if it was a pcb, a psu or the transistor.  A DMM measurement from display ground to psu ground showed 50 ohms, which seems high.  On the new setup, I took the display ground straight to the psu ground.  Anyhow, I'm all set now, project is working and looking good.

closed 2020-09-08 by exeljb