aderusha/HASwitchPlate #122: Wemos D1 mini availability

bgardner46, 2020-09-14:
I built my first HASP and, after writing a short TCL plugin, it works great with my Homevision system. I would like to build a few more but have a question about sourcing more d1 minis. The link in the documentation now points to a device called a d1 mini nodemcu and I'm not sure if they are physically or electrically compatible. I did see the link below on Amazon but it is v3.0.0 and the ones I purchased from Aliexpress that are currently working are 3.1.0. Aliexpress lists these as no longer available.
[https://www.amazon.com/Stemedu-Arduino-Nodemcu-MicroPython-Female/dp/B0894XL9TW/ref=sr_1_10?dchild=1&keywords=wemos+d1+mini&qid=1600116407&s=electronics&sr=1-10](url)

Any suggestions?
Thanks for this great project and for sharing it with all of us.
Bob

aderusha, 2020-09-14:
While I've not tested all the various D1 mini versions out there, I've never had a particular version not work. [Their info page](https://docs.wemos.cc/en/latest/d1/d1_mini.html) isn't really clear on what is changing between revisions, but I think it's safe to expect it's something regarding component availability over time.  From our perspective, all the various versions should work.

So - I think you should be all set!

Have you published your work on the Homevision integration anywhere?  Really curious to hear more about that!

bgardner46, 2020-09-14:
Okay, thanks. I'll try the V3.0.0 models and let you know if there's any
problem.

I haven't published anything yet but will add a page to my website soon and
let you know when it's available.

Bob

closed 2020-11-05 by aderusha