aderusha/HASwitchPlate #124: Real bad contrast for the display?

psarossy, 2020-11-05:
I got one of the pre-assembled units to test this design out (and I love every other part of it), but the screen seems to have horrific viewing angles from the top, where it's viewed from 99% of the time, and fine from the bottom.

Is this normal? Is the screen defective, or should be upside down?

https://photos.app.goo.gl/fy1NVs8dXJzjTkKT7

aderusha, 2020-11-05:
Nextion has changed their LCD panels around some time a couple years ago and the viewing angles aren't always great from some directions.  If the viewing angle works better from below, there is a simple (if somewhat stupid) fix - physically mount the device upside down, and then flash the Nextion firmware [HASwitchPlate-Inverted.tft](https://github.com/aderusha/HASwitchPlate/blob/master/Nextion_HMI/HASwitchPlate-Inverted.tft) to the LCD.  That firmware is the same as the normal LCD firmware, just flipped 180°.

Give it a shot and let me know how it goes!

psarossy, 2020-11-25:
Sorry for the delayed response... The inverted firmware and turning upside down fixed my issue, but since I used a 3 slot decora plate, I couldn't just mount it upside down, but had to print a new plate as the LCD is not symmetrical.

closed 2020-11-25 by psarossy