aderusha/HASwitchPlate #125: buttons are working but display is not getting the text

dimitrybradt, 2020-11-10:
Hello,

While the text buttons are not being displayed on the device, the d1 receives the commands from HA.
I've used an NX4832T035 at this point since I'm still waiting for the delivery of the NX3224T024.
Should I do anything more (than enlarging the buttons) in the nextion editor for a fix?

Thanks!

[image]

aderusha, 2020-11-10:
Nearly 100% this is a wiring problem.  Check your solder joints, this is what happens when TX from the panel doesn't make it to D7 on the Wemos D1.  You should be able to check w/ a continuity test between those pins.

dimitrybradt, 2020-11-10:
Hi,

The setup is still with Dupont cables...  I'm getting serial connection with display and Arduino board. so that's not the issue.
I must say if I use the emulator it doesn't show me a lot either (based on the code of output of the board.)
firmware issue? I'm using latest haspplate firmware and stock screen firmware but loaded correct nextion display through a serial connection.

aderusha, 2020-11-10:
Based on your log, you are not getting two-way communication with your LCD.  The firmware for both devices is in pretty wide use at this point, and while it's certainly not perfect it does work.  You will be best served to confirm a full signal path between the Wemos and the LCD, because that's almost certainly where your problem lies.

aderusha, 2020-12-19:
Following up - did you get your HASP working as expected?

closed 2020-12-23 by aderusha