aderusha/HASwitchPlate #128: Corrupt or incomplete STL file

mike-sherman, 2021-01-02:
The file *HASwitchPlate/3D_Printable_Models/HASwitchPlate_desktop_base.stl* is not a valid 3D object.
OpenSCAD will not load it and it shows as below in Prusa Slicer and in MeshLab (attached).
[image]

aderusha, 2021-01-04:
Is there any chance you might have a corrupted file on your side?  I've just gone and downloaded the STL as published on the repo, and it shows clean in both PrusaSlicer and OpenSCAD:

[image]

[image]

To the best of my knowledge all the models published here should be manifold.

closed 2021-01-30 by aderusha

mike-sherman, 2021-02-07:
Downloaded again and it is OK. 
Sorry.
-Mike