aderusha/HASwitchPlate #130: Alarm

josheharvey, 2021-01-15:
I'm having an issue where if the alarm is "pending" or "triggered" the HASP will not allow for it to be disabled. Only if it is "away_home" mode. This isn't useful because the alarm will switch to "pending" when motion is detected but because of that, I'm not able to disable the alarm via the HASP. I've tried adding it as a condition but can't seem to get it to work

Any help would be appreviated!

IIAIronWolf, 2021-03-02:
I've noticed for mine, I cannot disarm the alarm while the alarm is still pending. Once the system is armed however, I can type in the code and disarm.

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha