aderusha/HASwitchPlate #131: deployhasp.sh breaks configuration.yaml when he file doesn't end with a new line

mpoli, 2021-01-22:
When the last character of configuration.yaml isn' t a new line (line feed), it will append "recorder:" to the last line in the configuraion.yaml file, and HA will accuse that configuration.yaml is invalid because of some block error, mine was as follows:

>Error loading /config/configuration.yaml: while parsing a block mapping
>in "/config/configuration.yaml", line 188, column 11
>expected <block end>, but found '<scalar>'
>in "/config/configuration.yaml", line 188, column 20

Quick fix: hit an enter before the " recorder:" and put it alone in a new line.

aderusha, 2021-01-22:
Fix is in `dev`, but it's probably better if I pushed that part into `master` while I work on the next release.  I've gone and done that, thanks for the heads up!

closed 2021-01-22 by aderusha

mpoli, 2021-01-24:
@aderusha Congrats on the excellent project!