aderusha/HASwitchPlate #135: Configuration Saved, but immediately forgotten

mpoli, 2021-02-20:
I am having a persistent issue with 0.40 that causes the configuration, expect for SSID and SSID password to be "forgotten"  or not "sitck", and even the board name reverts to plate01 when I follow this procedure:

Recorded with ESP8266 flasher, DOUT, 1MB config

I boot the recently recorded device and I will in SSID and SSID password, which is 50+ chars long.

After the reboot, the device connects normally to the Wi-Fi network.

I open the web interface and start filling up the remaining information:

- HASP node name is 15 chars, all lower case and one "_" 
- MQTT broker is an IP address
- MQTT port left at the default 1883
- MQTT user: "mqtt" 
- MQTT password: a mix of lower case, upper case and special chars and numbers, 10+ chars long
- HASP Admin name left to default
- HASP Admin Password is 12 chars, upper case and lower case

Remaning options set to default.

I hit " save settings" 

A screen with the correct name I just input pops up with:

caretaker_plate

Saving updated configuration values and restarting device

But then, nothing is really saved. The next time I open the screen, only the Wifi information is preserved, everything else reverts to default and I get this information from the from part of the screen:

MQTT Status: Disconnected, return code: 0
MQTT ClientID:
HASP Version: 0.40
LCD Model:
LCD Version: 0
LCD Active Page: 0
CPU Frequency: 160MHz
Sketch Size: 477536 bytes
Free Sketch Space: 2666496 bytes
Heap Free: 20176
Heap Fragmentation: 1
ESP core version: 2_6_3
IP Address: 192.168.1.76
Signal Strength: -69
Uptime: 15
Last reset: Fatal exception:0 flag:4 (SOFT_RESTART) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000

alexives, 2021-03-04:
I had the same problem, but I tried the procedure here - https://github.com/aderusha/HASwitchPlate/issues/107#issuecomment-629784869 and it seems to have worked for me.

> re-flash the firmware through the admin web page, followed by a factory reset.

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha