aderusha/HASwitchPlate #137: Backlight MQTT message ignored if page set payload is first/same time.

marc-gist, 2021-03-01:
I have an automation that worked before the update to the latest beta/dev branch bin as of 11 days ago.
My Screen now flickers when changing pages, no big deal.  However, what broke on this update is>

When my lights turn off, I send two payloads to the screen (which worked before).

`hasp/plate01/brightness/set = 10`
`hasp/plate01/command/page = 2`

Basically at the same time.  If I add a slight delay to the Brightness one, it still does now work (brightness doesn't change).
If I add a delay to the Page (so brightness gets in first for sure), it works fine.

I've not change anything else, just the firmware. so not sure what is causing this.

aderusha, 2021-03-01:
I'm unable to re-create this behavior.

```bash
mosquitto_pub -h hassio -u hassio -P hassio -t 'hasp/plate01/brightness/set' -m '10'
mosquitto_pub -h hassio -u hassio -P hassio -t 'hasp/plate01/command/page' -m '2'
```

Works as expected on the current build.  I'll keep testing this as work continues.

marc-gist, 2021-03-01:
> I'm unable to re-create this behavior.
> 
> ```shell
> mosquitto_pub -h hassio -u hassio -P hassio -t 'hasp/plate01/brightness/set' -m '10'
> mosquitto_pub -h hassio -u hassio -P hassio -t 'hasp/plate01/command/page' -m '2'
> ```
> 
> Works as expected on the current build. I'll keep testing this as work continues.

Thanks for testing. I will see if I can find an issue in my automations.

aderusha, 2022-09-06:
Closing as this repo is not being maintained.  [Check here for the updated project](https://github.com/HASwitchPlate/HASPone)!

closed 2022-09-06 by aderusha