aderusha/HASwitchPlate #138: Desktop USB cutout line up

bgardner46, 2021-03-02:
I wasn't sure if this was the correct place to address this but here goes. I have built several in wall units and they work great- you do great work and I am impressed. I decided to build a desktop unit and the parts printed great but I can't figure out how to get the USB port on the d1mini to line up with the cutout in the back and the stand. Maybe I'm missing something but could use some suggestions for mounting the mini. I actually modified the back by making the hole a little larger and then mounted the d1mini at an angle and it sort of works but I feel like I'm missing something.

Thanks again for a great project.

Bob

aderusha, 2021-03-02:
Are you assembling this on the PCB?  Can you post pics of what your board looks like?

bgardner46, 2021-03-02:
I'm assembling on the same PCB as the wall units. I'm mounting the
d1mini directly to the PCB with standard headers. Here's a top view of the
original back:

[image: topview.jpg]

and here's a side view:

[image: sideview.jpg]

here's a finished unit with the modified d1mini install:

[image: desktop.jpg]

So any suggestions? Am I doing something wrong?

Thanks,

Bob

aderusha, 2021-03-02:
Looks like the photos aren't working in your post.  Can you maybe drop them on imgur or similar?

bgardner46, 2021-03-02:
How about a Dropbox share link?

https://www.dropbox.com/sh/wi7o6gd5bsj910x/AABJnZ58sI9hFjtXRzadKY-Ea?dl=0

bgardner46, 2021-03-03:
Here's an Imgur link as well. I set them as hidden but assume that you can
still view them with a link - not a regular Imgur user.

https://imgur.com/a/GOv8wAn

Bob

aderusha, 2021-03-03:
Not sure if this is the problem but it appears that you're using the newer D1 mini board, [the BOM link](https://amzn.to/2UZlga4) has the right device which is a slightly different from the D1 you have in the photos above.

Specifically, it appears the version of the board you are using has the USB micro port soldered to the top instead of the bottom, and everything is out of alignment as a result.

This is pretty annoying but it's clearly the result of me under-specifying the correct parts to use :(

bgardner46, 2021-03-03:
I think you nailed it. I didn't even realize there were boards with the USB
connector on the bottom - my bad. I just ordered a couple.

Thanks for the quick response.

Bob

closed 2021-03-04 by bgardner46